sparkfun/nRF52840_Breakout_MDBT50Q #7: Serial1 Definition Missing

Phando, 2020-05-22:
Hello,
I am just getting started with the nrf52840. The variant does not define Serial1 as specified in the Arduino setup doc

`
Hardware Serial (Serial1) to pins 17 (TX) and 15 (RX) to match the Arduino Pro Mini pinout
`

Should I just double declare Serial1?

```
#define PIN_SERIAL_RX       (15)
#define PIN_SERIAL_TX       (17)
#define PIN_SERIAL1_RX      (15)
#define PIN_SERIAL1_TX      (17)
```

bboyho, 2020-05-29:
Hi,

Interesting... It looks like the nRF52 Arduino core files were updated with some additional features. This update broke a few things related to the UART sometime between [**0.17.0**](https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/0.17.0/cores/nRF5/Uart.cpp#L243) and [**0.16.0**](https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/0.16.0/cores/nRF5/Uart.cpp#L207). This is probably what you are seeeing with the v0.20.0 release:

    C:\Users\user_bob\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.20.1\cores\nRF5\Uart.cpp:247:48: error: 'PIN_SERIAL1_RX' was not declared in this scope
    
       Uart Serial1( NRF_UARTE0, UARTE0_UART0_IRQn, PIN_SERIAL1_RX, PIN_SERIAL1_TX );
    
                                                    ^~~~~~~~~~~~~~
    
    C:\Users\user_bob\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.20.1\cores\nRF5\Uart.cpp:247:48: note: suggested alternative: 'PIN_SERIAL_RX'
    
       Uart Serial1( NRF_UARTE0, UARTE0_UART0_IRQn, PIN_SERIAL1_RX, PIN_SERIAL1_TX );
    
                                                    ^~~~~~~~~~~~~~
    
                                                    PIN_SERIAL_RX
    
    C:\Users\user_bob\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.20.1\cores\nRF5\Uart.cpp:247:64: error: 'PIN_SERIAL1_TX' was not declared in this scope
    
       Uart Serial1( NRF_UARTE0, UARTE0_UART0_IRQn, PIN_SERIAL1_RX, PIN_SERIAL1_TX );
    
                                                                    ^~~~~~~~~~~~~~
    
    C:\Users\user_bob\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.20.1\cores\nRF5\Uart.cpp:247:64: note: suggested alternative: 'PIN_SERIAL_TX'
    
       Uart Serial1( NRF_UARTE0, UARTE0_UART0_IRQn, PIN_SERIAL1_RX, PIN_SERIAL1_TX );
    
                                                                    ^~~~~~~~~~~~~~
    
                                                                    PIN_SERIAL_TX
    
    exit status 1
    Error compiling for board SparkFun Pro nRF52840 Mini.

For backward compatibility, it would not hurt to double declare the hardware serial pins. Testing this out with **0.16.0** and **0.20.1** (which is the latest as of today), it did not seem to cause any issues. I was able to still send serial data via Arduino monitor using `Serial.print()`. Using a serial-to-USB converter, I was still able to send/receive serial data via hardware serial pins using  `Serial1.print()`. I'll adjust the **variants.h** file in this repo in case others run into the same issue. 

Here's the gist of it if you (or anyone else that comes across this issue)  want to test it out => https://gist.github.com/bboyho/554cdcdac422a8cf07f4f2d870120d38 .

bboyho, 2020-05-29:
And it's been added to this GitHub repo. =)

closed 2020-05-29 by bboyho