sparkfun/nRF52840_Breakout_MDBT50Q #9: Bootloader Firmware Source?

cHemingway, 2022-05-17:
The firmware link in the [product page](https://www.sparkfun.com/products/15025) just links to the [adafruit repository](https://github.com/adafruit/Adafruit_nRF52_Bootloader), which does not contain a board definition for the nRF52840 board, and so will not work properly without a board definition being created (wrong name shows up in device manager, wrong pins for LEDs/Buttons etc).

Can a board definition be provided? And (ideally) merged into the Adafruit repo?

cHemingway, 2022-05-17:
I have created a new board ` boards\sparkfun_pro_nrf52840_mini ` based on  `boards\sparkfun_nrf52840_micromod`, and it seems to work.
https://gist.github.com/cHemingway/26a6dba63699b8397ca0682419dc22db

However, I haven't changed the VID/PID from the micromod version, I think each board should have a unique VID/PID

brhoff720, 2022-05-17:
This board should have the following two PID 0x5284 and 0x5285.  SparkFun's VID is 0x1B4F.

cHemingway, 2022-05-20:
Thanks @brhoff720! Do you work for either Adafruit or Sparkfun? If so, are you interested in getting support for this board merged into the Adafruit repo?

A concern I have with submitting a pull request to the Adafruit repo myself, is that that doesn't gaurantee the bootloader programmed on a board when you buy it is actually running that particular code. It would be useful to know what the sparkfun board is actually running

makin-stuff, 2022-05-24:
Hello @cHemingway . 
Have you seen this [article by SparkFun](https://learn.sparkfun.com/tutorials/nrf52840-development-with-arduino-and-circuitpython)? It goes into all the details of getting the [Pro nRF52840 Mini](https://www.sparkfun.com/products/15025) set up in the Arduino IDE using the Adafruit core. 
Newer products like the [MicroMod nRF52840 Processor Board
](https://www.sparkfun.com/products/16984) use the [Arduino MBed Core](https://learn.sparkfun.com/tutorials/micromod-nrf52840-processor-hookup-guide/arduino-software-setup) with custom boards.txt file and variants for the SF product.

closed 2022-05-24 by makin-stuff

cHemingway, 2022-05-24:
Hi @makin-stuff 

I don't think this is completed, as I don't have the source code for the bootloader, which was my question. I am interested in it as it sets things like the Nordic Softdevice used.
Instead, you've pointed me at some other things, which I don't think are related?

- An article on how to use CircuitPython and Arduino, which isn't the bootloader source. Instead, the article just says the board _uses_ a bootloader, but doesn't tell you anything else about it.

- Information on other boards, which I'm not using

- Boards.txt, which is Arduino (not CircuitPython), and isn't what I'm looking for.

If Sparkfun don't want to share the source code, that would be an answer to my question, though not my preferred answer of course!

makin-stuff, 2022-05-25:
Alrighty @cHemingway. 

I did some searching and I don't think we have the bootloader source firmware for this product and the original engineer no longer works at SparkFun. It is likely that one of the feather board definitions (feather_nrf52840_express or feather_nrf52840_sense) from this [repo](https://github.com/adafruit/Adafruit_nRF52_Bootloader) was used to generate the binaries we use to upload to the SF Pro nRF52840 Mini and he never created a pull request to add those defs to the Adafruit repo.

Additionally, one of our production steps is uploading this [softdevice hex file](https://github.com/NordicSemiconductor/nRF5-SDK-for-Mesh/blob/master/bin/softdevice/s140_nrf52_6.1.1_softdevice.hex) from this [repo.](https://github.com/NordicSemiconductor/nRF5-SDK-for-Mesh)

I also double-checked the PID/VID's given by @brhoff720 and those are correct!

Sorry I couldn't be of more help.

cHemingway, 2022-05-26:
@makin-stuff 

OK thanks, I appreciate you chasing this one down! 
Maybe the product page could be updated with a link to the binaries instead?